jjhelmus/berryconda #1: xlrd package

capyy, 2017-03-05:
Hello Jonathan,

I tried to import Excel data in Pandas (with : pd.read_excel) and got :
ImportError: No module named 'xlrd'

xlrd is not in your repo and also tried from conda but no success :
PackageNotFoundError: Package not found: '' Package missing in current linux-armv7l channels: 
  - xlrd

Thanks !!
(Thanks for you very useful repo on a Pi !! great time saver)

jjhelmus, 2017-03-08:
I have not yet built xlrd for the Raspberry Pi.  It looks like conda-forge has a [xlrd feedstock](https://github.com/conda-forge/xlrd-feedstock) which can serve as a source of the rpi recipe.  My collections of Pi's that I use for building packages are getting some updates (new SD cards!) so it might take a few days for me to get to this.  I'll add to this issue with updates.

jjhelmus, 2017-03-10:
I built the `xlrd` package and uploaded to the `rpi` channel for Python 2.7, 3.5 and 3.6.  Both armv6l and armv7v package are available. You will need to install this in addition to `pandas` as it is not a required dependency.

closed 2017-03-10 by jjhelmus