jjhelmus/berryconda #4: Pip under berryconda behaves differently than in does under anaconda.

Michael-F-Ellis, 2017-04-24:
First. thanks very much for creating this package!

Next: How do I make pip play nice with berryconda? I have a project on a Pi3 that needs setproctitle. It's not available thru conda.  On OS X using Anaconda I'm used to being able to install packages with pip and have all the paths come out correctly in the virtual environment. 

The terminal session below shows some system info and my path in a Python 3.5 root environment. Then I switch to a Python 2.7 environment and show the path.  Trying `pip install setproctitle` says it's already available under /usr/local/lib.

Then I start Python, check sys.path and then try to import setproctitle but get an ImportError -- which makes sense because sys.path only contains the pwd and the berryconda paths.

What's the recommended fix?

Thanks,
Mike

```
py27) pi@pi2:~ $ uname -a
Linux pi2 4.4.34-v7+ #930 SMP Wed Nov 23 15:20:41 GMT 2016 armv7l GNU/Linux

pi@pi2:~ $ echo $PATH
/home/pi/berryconda3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games

pi@pi2:~ $ source activate py27
(py27) pi@pi2:~ $ echo $PATH
/home/pi/berryconda3/envs/py27/bin:/home/pi/berryconda3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
(py27) pi@pi2:~ $ which pip
/usr/local/bin/pip

(py27) pi@pi2:~ $ pip install setproctitle
Requirement already satisfied: setproctitle in /usr/local/lib/python2.7/dist-packages

(py27) pi@pi2:~ $ python
Python 2.7.13 | packaged by rpi | (default, Jan 15 2017, 19:52:30) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.path
['', '/home/pi/berryconda3/envs/py27/lib/python27.zip', '/home/pi/berryconda3/envs/py27/lib/python2.7', '/home/pi/berryconda3/envs/py27/lib/python2.7/plat-linux2', '/home/pi/berryconda3/envs/py27/lib/python2.7/lib-tk', '/home/pi/berryconda3/envs/py27/lib/python2.7/lib-old', '/home/pi/berryconda3/envs/py27/lib/python2.7/lib-dynload', '/home/pi/berryconda3/envs/py27/lib/python2.7/site-packages']

>>> import setproctitle
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named setproctitle
>>> 
```

jjhelmus, 2017-04-24:
@Michael-F-Ellis It looks like the system `pip` is getting called and `setproctitle` is getting installed into the system Python's site-packages directory.  Try install pip into the conda environment using `conda install pip` and then installing `setproctitle`.  You can also check which `pip` is getting called with `which pip`.  To install packages into the conda environment the path should include the berryconda directory.

Michael-F-Ellis, 2017-04-24:
Thanks for the quick response! Forgot to mention that I tried that earlier.  Tried it again.  Here's what happens:
```
(py27) pi@pi2:~ $ conda install pip
Fetching package metadata .....
Solving package specifications: ..........

Package plan for installation in environment /home/pi/berryconda3/envs/py27:

The following NEW packages will be INSTALLED:

    certifi:    2017.4.17-py_0
    packaging:  16.8-py27_0   
    pip:        9.0.1-py27_0  
    pyparsing:  2.2.0-py27_0  
    setuptools: 35.0.1-py27_0 
    six:        1.10.0-py27_0 
    wheel:      0.29.0-py27_0 

Proceed ([y]/n)? y

Linking packages ...
Installing certifi-2017.4.17-py_0 requires a minimum conda version of 4.3.
(py27) pi@pi2:~ $ which pip
/usr/local/bin/pip  
```

Michael-F-Ellis, 2017-04-24:
Thought for a moment I had it fixed by going to my root python3.5 environment and conda installing pip there. That succeeded and I got setproctitle to install and import in python 3.5.  But then I switched back to 2.7 to try things there.  I was able to get conda install pip to succeed and pip install setproctitle now finds in the berryconda 3.5 site package. But it still won't import into py 2.7

```
(py27) pi@pi2:~ $ conda install pip
Fetching package metadata .....
Solving package specifications: .

Package plan for installation in environment /home/pi/berryconda3/envs/py27:

The following NEW packages will be INSTALLED:

    certifi:    2017.4.17-py_0
    packaging:  16.8-py27_0   
    pip:        9.0.1-py27_0  
    pyparsing:  2.2.0-py27_0  
    setuptools: 35.0.1-py27_0 
    six:        1.10.0-py27_0 
    wheel:      0.29.0-py27_0 

Proceed ([y]/n)? y

(py27) pi@pi2:~ $ which pip
/home/pi/berryconda3/envs/py27/bin/pip
(py27) pi@pi2:~ $ pip install setproctitle
Requirement already satisfied: setproctitle in ./berryconda3/lib/python3.5/site-packages
(py27) pi@pi2:~ $ python
Python 2.7.13 | packaged by rpi | (default, Jan 15 2017, 19:52:30) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import setproctitle
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named setproctitle
```

Michael-F-Ellis, 2017-04-24:
Got it working by a bit of trickery. 
1.  In the root environment, I moved the setproctitle .so and dist-info files to /tmp
2. Switched to( py27) and did `pip install setproctitle`. This time it worked.
3. Verified it would now import.
4. Switched back to (root)
5. Moved files back from /tmp to /berryconda3/lib/python3.5/site-packages/
6. Verified that setproctitle imports ok in (root)

Hope you can find a better solution.

Cheers,
Mike

jjhelmus, 2017-04-25:
Odd, I tried to replicate this and was able to installing pip and then setproctitle into a python 2.7 conda environment without issue.  
I wonder if something is pip's caching logic was causing a previous build of setproctitle to be found when installing into one of the environments?  In the future it might be helpful to see if clearing the cache helps, `rm -rf ~/.cache/pip` will do that, or add `--no-cache-dir` to the pip install command.  Other than that I'm not sure what else to try, but I'm glad you found a solution.

I can create `setproctitle` conda packages without to much effort if that would help.

Michael-F-Ellis, 2017-04-25:
I think you're probably right about the cache.  It kind of makes sense given that the PI already had a number of pip-installed packages including setproctitle before I installed berryconda for Python 3.

Setproctitle is useful in my work on control systems that have multiple processes running. I use it to apply a common tag so I can use ps and pkill to inspect and control them from the command line.  Not sure many other people use it, so packaging it might not be the best leverage for your time.  I do appreciate the offer, though, and would certainly use it if it were available.

Cheers,
Mike

jjhelmus, 2017-04-26:
I created packages for `setproctitle` and uploaded them to the `rpi` channel.  In the future you should be able to install the package using `conda install setproctitle` rather than using pip.

closed 2017-04-26 by jjhelmus

Michael-F-Ellis, 2017-04-26:
Awesome. Thanks!