jjhelmus/berryconda #5: bin/conda being removed

nishadhka, 2017-05-11:
Thank you very much for Berryconda.

There is a problem in installing ```conda install -c rpi conda-build=2.1.10``` in RPi B, having raspbian jessie 8 with Berryconda2-1.0.0-Linux-armv7l.sh. The error is as follows 
```
ERROR conda.core.link:_execute_actions(335): An error occurred while installing package 'defaults::conda-build-2.1.10-py27_0'.
IOError(2, 'No such file or directory')
Attempting to roll back.
IOError(2, 'No such file or directory')
```
More than the error, further try for the conda command ends in ```-bash: /home/pi/berryconda2/bin/conda: No such file or directory```, making to suspect the installation command completely removes the conda bin from the berryconda installation. To get back the lost bin/conda, a fresh installation of the berryconda is followed. Similar losing of bin/conda was observed for installing packages such as anaconda-client etc too.

jjhelmus, 2017-05-11:
@nishadhka Thanks for posting this bug, sorry that updating conda-build broken your install.  

It looks like a number of packages in the `rpi` channel are including an extra `bin/conda` file due to a bug in conda-build (conda/conda-build#1912).  I'll scan through the repo and try to identify the affected packages and replace them with with new packages. 

If you want to fix your conda install it should be possible to unpack a conda bz2 package directly into the install location.  There is likely a copy in your installs pkgs directory.

jjhelmus, 2017-05-12:
I've gone through the packages in the `rpi` channel looking for any with an extra `bin/conda` file.  From my search it looks like the only offending packages are those for `conda-build` version2.1.10.  I've labeled those packages as `broken` so they are not installed by default and am replacing them with a new build using a version of conda-build which does not include this extra file.

closed 2017-05-12 by jjhelmus

nishadhka, 2017-05-13:
Thank you very much.

nishadhka, 2017-05-16:
new attempt with anaconda-client and conda-build, still shows the behaviour, it seems with the sh file having the defective conda version. 

such as with ```conda install anconda-client``` ends in ```Installing pytz-2017.2-py_0 requires a minimum conda version of 4.3.``` and ```conda install conda-build``` ends in ```Installing beautifulsoup4-4.6.0-py_0 requires a minimum conda version of 4.3.```

However, a conda upgrade with ```conda upgrade conda``` avoids bin/conda removal

nishadhka, 2017-05-16:
The conda upgrade solves the issue, a new build of [pybluezv0.22](https://anaconda.org/ueic/pybluez) could be made using berryconda, thank you.

commit bd5e6339 (2017-05-12): Bump build number on conda-build recipe

Release new conda-build packages which do not contain
an extra /bin/conda file.

closes #5