jjhelmus/berryconda #6: Package Request - pandas datareader (Berry Conda on OSMC)

SystemTheory, 2017-05-12:
I have a package build request for pandas-datareader so it will be available on the rpi channel (per the Anaconda docs here):

https://anaconda.org/anaconda/pandas-datareader

Also regarding compatibility you may wish to know that I successfully installed Berry Conda on OSMC, after installing bzip2, as described in this thread:

https://discourse.osmc.tv/t/can-i-install-berry-conda-python-package-manager-on-osmc/36630

Thank you for any attention you may be able to devote to this package request.

jjhelmus, 2017-05-13:
@SystemTheory Great to hear that Berryconda works on OSMC!

I've went ahead and built `pandas-datareader` packages for armv6l and armv7l and uploaded them to the `rpi` channel.  You should be able to install them using `conda install pandas-datareader`.  Note that conda may need to downgrade your version of `pandas` as `pandas-datareader` does not yet support `pandas` 0.20.x.

closed 2017-05-13 by jjhelmus

jjhelmus, 2017-05-20:
Thanks for the feedback on the install process.  I've update the README file with some text mentioning that bzip2 should be installed and a suggestion about installing in a user account and not root or sudo.