jjhelmus/berryconda #8: Jupyter in Berryconda - Native kernel (python3) is not available

DroneMan, 2017-06-09:
There appears to be an issue running Jupyter notebook in Berryconda.

I'm using a Raspberry Pi 3 Model B, with Raspbian Jessie 8. Using Berryconda3-1.0.0-Linux-armv7l.sh.

Typing jupyter run or jupyter-run yields the following error:

```
pi@raspberrypi:~ $ jupyter-run
[RunApp] WARNING | Native kernel (python3) is not available
Traceback (most recent call last):
  File "/home/pi/berryconda3/lib/python3.5/site-packages/jupyter_client/kernelspec.py", line 201, in get_kernel_spec
    resource_dir = d[kernel_name.lower()]
KeyError: 'python3'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/berryconda3/bin/jupyter-run", line 6, in <module>
    sys.exit(jupyter_client.runapp.main())
  File "/home/pi/berryconda3/lib/python3.5/site-packages/jupyter_core/application.py", line 267, in launch_instance
    return super(JupyterApp, cls).launch_instance(argv=argv, **kwargs)
  File "/home/pi/berryconda3/lib/python3.5/site-packages/traitlets/config/application.py", line 657, in launch_instance
    app.initialize(argv)
  File "<decorator-gen-7>", line 2, in initialize
  File "/home/pi/berryconda3/lib/python3.5/site-packages/traitlets/config/application.py", line 87, in catch_config_error
    return method(app, *args, **kwargs)
  File "/home/pi/berryconda3/lib/python3.5/site-packages/jupyter_client/runapp.py", line 76, in initialize
    JupyterConsoleApp.initialize(self)
  File "/home/pi/berryconda3/lib/python3.5/site-packages/jupyter_client/consoleapp.py", line 334, in initialize
    self.init_kernel_manager()
  File "/home/pi/berryconda3/lib/python3.5/site-packages/jupyter_client/consoleapp.py", line 288, in init_kernel_manager
    self.kernel_manager.start_kernel(**kwargs)
  File "/home/pi/berryconda3/lib/python3.5/site-packages/jupyter_client/manager.py", line 244, in start_kernel
    kernel_cmd = self.format_kernel_cmd(extra_arguments=extra_arguments)
  File "/home/pi/berryconda3/lib/python3.5/site-packages/jupyter_client/manager.py", line 175, in format_kernel_cmd
    cmd = self.kernel_spec.argv + extra_arguments
  File "/home/pi/berryconda3/lib/python3.5/site-packages/jupyter_client/manager.py", line 87, in kernel_spec
    self._kernel_spec = self.kernel_spec_manager.get_kernel_spec(self.kernel_name)
  File "/home/pi/berryconda3/lib/python3.5/site-packages/jupyter_client/kernelspec.py", line 203, in get_kernel_spec
    raise NoSuchKernel(kernel_name)
jupyter_client.kernelspec.NoSuchKernel: No such kernel named python3
```

jjhelmus, 2017-06-13:
@DroneMan I believe this error can occur when the `ipykernel` or `notebook` package is missing from the environment.  To run Jupyter notebooks you need to install the `notebook` package not the `jupyter_core` or `jupyter_client` packages, although these will be installed. Can you see if `conda install notebook` fixes this issue?  If it does not the output of `conda list` would help debug the issue?

DroneMan, 2017-06-15:
Thanks, installing notebook did fix the issue!

closed 2017-06-15 by DroneMan