jjhelmus/berryconda #10: Implement wxPython

DroneMan, 2017-06-15:
Any chance you could implement wxPython?

I'm interested in creating some recipes for other Python packages to help add to the rpi channel,  but I figured I'd throw in this request while I am learning to do so. Thanks!

DroneMan, 2017-06-15:
While this issue is still open, implementing pyUSB would be really fantastic as well.

jjhelmus, 2017-06-16:
I'm not planning on expanding the collection of packages on the `rpi` channel unless the packages are core PyData or Scientific Python packages.  Additionally from what I've been able to tell both `wxPython` and `pyUSB` are quite challenging builds.  

If you wanted to build these yourself and upload them to your own account on anaconda.org that would be a good solution.

closed 2017-06-16 by jjhelmus