jjhelmus/berryconda #12: `pip install -e` installs the source under the current working directory instead of under the environment directory

ksze, 2017-07-31:
I'm used to the behaviour of pip in the system-provided python and virtualenv/virtualenvwrapper, where `pip install -e` a package from git would clone the repo into a subdirectory of the virtualenv. For instance, using virtualenvwrapper, I can do this:

    $ workon foobarenv
    $ pip install -e git+https://github.com/foobar/foobar.git

And foobar's git repo would be cloned into `/home/ksze/.virtualenvs/foobar/src/foobar`, regardless of where I run pip from.

However, using berryconda, `pip install -e` just clones to a subdirectory under the _current working directory_. This behaviour is quite annoying, not to mention being incompatible with what people expect when using a requirements file.

I'm not sure if this is an upstream conda issue or specific to berryconda, as I have never used the "vanilla" conda.

ksze, 2017-07-31:
This is what `pip install --help` has to say about installing editable projects:

    --src <dir>                 Directory to check out editable projects into. The default in a virtualenv is "<venv path>/src". The default for global installs is "<current dir>/src".

I also made sure that I'm running pip in the conda environment:

    ~ $ which pip
    /home/ksze/berryconda3/envs/foobar/bin/pip

This suggests that pip somehow doesn't recognize a conda environment as a proper virtualenv.

However, you would probably agree that for all intents and purposes, a conda environment activated by `source activate foobar` should be treated like a plain virtualenv.

jjhelmus, 2017-07-31:
Unfortunately conda environments are not virtualenv and therefore `pip install -e` results in the default behavior of installing into the current working directory. I agree that it would be nice if conda and virtualenv were more aware of each other. This is not an issue specific to berryconda, the same behavior is seen when using conda on other platforms.  Opening issues upstream with with [pip](https://github.com/pypa/pip), [virtualenv](https://github.com/pypa/virtualenv) or [conda](https://github.com/conda/conda) would be a better place to discuss this topic.

closed 2017-07-31 by jjhelmus

Chong-Jiao, 2018-06-27:
How can I uninstall it if I install it by "pip install -e ."?

mrjamescollins, 2018-12-14:
'pip uninstall <module name>'
you can get this name by doing a 'pip list'