jjhelmus/berryconda #13: Import Pandas throws up an error

bonehead, 2017-08-09:
Importing Pandas on the raspberry pi, the following error crops up

    `pi@raspberrypi:~/scripts $ python
     Python 3.6.1 | packaged by rpi | (default, Apr 20 2017, 19:35:19) 
     [GCC 4.9.2] on linux
     Type "help", "copyright", "credits" or "license" for more information.
     >>> import pandas
    Traceback (most recent call last):
      File "/home/pi/berryconda3/lib/python3.6/site-packages/pandas/__init__.py", line 26, in <module>
    from pandas._libs import (hashtable as _hashtable,
      File "/home/pi/berryconda3/lib/python3.6/site-packages/pandas/_libs/__init__.py", line 3, in <module>
        from .tslib import iNaT, NaT, Timestamp, Timedelta, OutOfBoundsDatetime
    ImportError: /home/pi/berryconda3/lib/python3.6/site-packages/pandas/_libs/tslib.cpython-36m-arm-linux-gnueabihf.so: file too short
    During handling of the above exception, another exception occurred:
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "/home/pi/berryconda3/lib/python3.6/site-packages/pandas/__init__.py", line 35, in <module>
        "the C extensions first.".format(module))
     ImportError: C extension: /home/pi/berryconda3/lib/python3.6/site-
     packages/pandas/_libs/tslib.cpython-36m-arm-linux-gnueabihf.so: file too short not built. If you want 
     to import pandas from the source directory, you may need to run 'python setup.py build_ext --inplace 
    --force' to build the C extensions first.
    >>>`

jjhelmus, 2017-08-09:
@bonehead Can you provide more details on the environment where you are seeing this error?  Specifically, what version of pandas and other packages are installed.  The output of `conda info` and `conda list` should provide these details.

I was unsuccessful when I attempted to replicate this error on my machine using pandas 0.20.3-py36_0, pandas imported without error using this setup.

bonehead, 2017-08-11:
@jjhelmus Please find attached the output 

    `pi@raspberrypi:/ $ conda info
    Current conda install:

               platform : linux-armv7l
          conda version : 4.3.22
       conda is private : False
      conda-env version : 4.3.22
    conda-build version : not installed
         python version : 3.6.1.final.0
       requests version : 2.18.1
       root environment : /home/pi/berryconda3  (writable)
    default environment : /home/pi/berryconda3
       envs directories : /home/pi/berryconda3/envs
                          /home/pi/.conda/envs
          package cache : /home/pi/berryconda3/pkgs
                          /home/pi/.conda/pkgs
           channel URLs : https://conda.anaconda.org/rpi/linux-armv7l
                          https://conda.anaconda.org/rpi/noarch
            config file : /home/pi/berryconda3/.condarc
             netrc file : None
           offline mode : False
             user-agent : conda/4.3.22 requests/2.18.1 CPython/3.6.1 Linux/4.4.50-v7+ debian/8.0 glibc/2.19    
                UID:GID : 1000:1000`

     pi@raspberrypi:/ $ conda list
    # packages in environment at /home/pi/berryconda3:
    #
    asn1crypto                0.22.0                   py36_0  
    attrs                     17.2.0                    <pip>
    Automat                   0.6.0                     <pip>
    beautifulsoup4            4.6.0                     <pip>
    bleach                    2.0.0                    py36_0  
    bs4                       0.0.1                     <pip>
    ca-certificates           2017.4.17                     0  
    certifi                   2017.4.17                py36_0  
    cffi                      1.10.0                   py36_0  
    chardet                   3.0.4                    py36_0  
    conda                     4.3.22                   py36_0  
    conda-env                 2.6.0                         0  
    constantly                15.1.0                    <pip>
    cryptography              1.9                      py36_0  
    decorator                 4.1.2                      py_0  
    entrypoints               0.2.3                    py36_0  
    html5lib                  0.999999999              py36_0  
    hyperlink                 17.3.0                    <pip>
    idna                      2.5                      py36_0  
    incremental               17.5.0                    <pip>
    ipykernel                 4.6.1                    py36_0  
    ipython                   6.1.0                    py36_0  
    ipython_genutils          0.2.0                      py_0  
    jinja2                    2.8                      py36_0  
    jsonschema                2.6.0                    py36_0  
    jupyter_client            5.1.0                    py36_0  
    jupyter_core              4.3.0                    py36_0  
    libffi                    3.2.1                         0  
    libgfortran               3.0.0                         0  
    libsodium                 1.0.10                        0  
    markupsafe                0.23                     py36_0  
    mistune                   0.7.4                      py_0  
    nbconvert                 5.2.1                      py_0  
    nbformat                  4.3.0                    py36_0  
    ncurses                   5.9                           0  
    notebook                  5.0.0                      py_0  
    numpy                     1.13.1                   py36_0  
    openblas                  0.2.19                        0  
    openssl                   1.0.2l                        0  
    packaging                 16.8                     py36_0  
    pandas                    0.20.3                   py36_0  
[... 12 lines cut ...]
    python-dateutil           2.6.1                      py_0  
    pytz                      2017.2                     py_0  
    pyzmq                     16.0.2                   py36_0  
    readline                  6.3                           0  
    requests                  2.18.1                   py36_0  
    ruamel_yaml               0.11.14                  py36_0  
    setuptools                36.0.1                   py36_0 
    simplegeneric             0.8.1                    py36_0  
    six                       1.10.0                   py36_0  
    sqlite                    3.15.0                        0  
    terminado                 0.6                      py36_0  
    testpath                  0.3.1                    py36_0  
    tk                        8.6.6                         0  
    tornado                   4.5.1                    py36_0  
    traitlets                 4.3.2                    py36_0  
    Twisted                   17.5.0                    <pip>
    urllib3                   1.21.1                   py36_0  
    wcwidth                   0.1.7                    py36_0  
    webencodings              0.5.1                      py_0  
    wheel                     0.29.0                   py36_0  
    xz                        5.2.2                         0  
    yaml                      0.1.6                         0  
    zeromq                    4.1.6                         0  
    zlib                      1.2.11                        0  
    zope.interface            4.4.2                     <pip>

jjhelmus, 2017-08-16:
I'm not seeing anything obvious that would be causing an issue.  Mixing `pip` and `conda` can cause some problems but in this case it does not seem to be the cause.  It's possible that `tslib.cpython-36m-arm-linux-gpuabihf.so` file was corrupted or truncated at some point.  Can you check the size of the file using `ls -l  /home/pi/berryconda3/lib/python3.6/site-
 packages/pandas/_libs/tslib.cpython-36m-arm-linux-gnueabihf.so`.  It should be 7001272, it is it not then something is amiss.  

Removing and reinstalling pandas may fix the issue, `conda remove pandas` followed by `conda install pandas` should do the trick.

bonehead, 2017-08-17:
I purged Berryconda and reinstalled and it did the trick, Pandas reinstalls fine.

closed 2017-08-17 by bonehead