jjhelmus/berryconda #14: Importing Numpy on Raspberry Pi Zero W throws an error

bonehead, 2017-08-11:
So I got a new Raspberry Pi Zero W and tried getting Berryconda to run on it. I encountered an error when trying to import Numpy although the install seemed to have run without issues.
Please find the error and outputs of **conda info** and **conda list** below:

    pi@raspberrypi:~/berryconda3/pkgs $ which python
    /home/pi/berryconda3/bin/python
    pi@raspberrypi:~/berryconda3/pkgs $ python
    Python 3.6.1 | packaged by rpi | (default, Apr 21 2017, 01:13:08) 
    [GCC 4.9.2] on linux
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import numpy
    Traceback (most recent call last):
      File "/home/pi/berryconda3/lib/python3.6/site-packages/numpy/core/__init__.py", line 16, in 
    <module>
        from . import multiarray
    ImportError: libgfortran.so.3: cannot open shared object file: No such file or directory

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "/home/pi/berryconda3/lib/python3.6/site-packages/numpy/__init__.py", line 142, in <module>
        from . import add_newdocs
      File "/home/pi/berryconda3/lib/python3.6/site-packages/numpy/add_newdocs.py", line 13, in 
     <module>
        from numpy.lib import add_newdoc
      File "/home/pi/berryconda3/lib/python3.6/site-packages/numpy/lib/__init__.py", line 8, in <module>
         from .type_check import *
      File "/home/pi/berryconda3/lib/python3.6/site-packages/numpy/lib/type_check.py", line 11, in 
    <module>
        import numpy.core.numeric as _nx
      File "/home/pi/berryconda3/lib/python3.6/site-packages/numpy/core/__init__.py", line 26, in 
    <module>
        raise ImportError(msg)
    ImportError: 
    Importing the multiarray numpy extension module failed.  Most
    likely you are trying to import a failed build of numpy.
    If you're working with a numpy git repo, try `git clean -xdf` (removes all
    files not under version control).  Otherwise reinstall numpy.

    Original error was: libgfortran.so.3: cannot open shared object file: No such file or directory
Output of **conda info**:

    pi@raspberrypi:~/berryconda3/pkgs $ conda info
    Current conda install:

               platform : linux-armv6l
          conda version : 4.3.22
       conda is private : False
      conda-env version : 4.3.22
    conda-build version : not installed
         python version : 3.6.1.final.0
       requests version : 2.18.1
       root environment : /home/pi/berryconda3  (writable)
    default environment : /home/pi/berryconda3
       envs directories : /home/pi/berryconda3/envs
                          /home/pi/.conda/envs
          package cache : /home/pi/berryconda3/pkgs
                          /home/pi/.conda/pkgs
           channel URLs : https://conda.anaconda.org/rpi/linux-armv6l
                          https://conda.anaconda.org/rpi/noarch
            config file : /home/pi/berryconda3/.condarc
             netrc file : None
           offline mode : False
             user-agent : conda/4.3.22 requests/2.18.1 CPython/3.6.1 Linux/4.9.35+ debian/8.0 glibc/2.19    
                UID:GID : 1000:1000

Output of **conda list**:

    pi@raspberrypi:~/berryconda3/pkgs $ conda list
    # packages in environment at /home/pi/berryconda3:
    #
    asn1crypto                0.22.0                   py36_0  
    beautifulsoup4            4.6.0                    py36_1  
    ca-certificates           2017.4.17                     0  
    certifi                   2017.4.17                py36_0  
    cffi                      1.10.0                   py36_0  
    chardet                   3.0.4                    py36_0  
    conda                     4.3.22                   py36_0  
    conda-env                 2.6.0                         0  
    cryptography              1.9                      py36_0  
    cython                    0.26                     py36_0  
    idna                      2.5                      py36_0  
    libffi                    3.2.1                         0  
    libgfortran               3.0.0                         0  
    ncurses                   5.9                           0  
    numpy                     1.13.1                   py36_0  
    openblas                  0.2.19                        0  
    openssl                   1.0.2l                        0  
    packaging                 16.8                     py36_0  
    pip                       9.0.1                    py36_0  
    pycosat                   0.6.1                    py36_0  
    pycparser                 2.17                     py36_0  
    pyopenssl                 16.2.0                   py36_0  
    pyparsing                 2.2.0                    py36_0  
    python                    3.6.1                         0  
    python-dateutil           2.6.1                      py_0  
    pytz                      2017.2                     py_0  
    readline                  6.3                           0  
    requests                  2.18.1                   py36_0  
    ruamel_yaml               0.11.14                  py36_0  
    setuptools                36.0.1                   py36_0  
    six                       1.10.0                   py36_0  
    sqlite                    3.15.0                        0  
    tk                        8.6.6                         0  
    urllib3                   1.21.1                   py36_0  
    wheel                     0.29.0                   py36_0  
    xz                        5.2.2                         0  
    yaml                      0.1.6                         0  
    zlib                      1.2.11                        0

jjhelmus, 2017-08-15:
I checked into this issue and it looks like the `openblas` package for armv6l (the Raspberry Pi 1 and Zero) incorrectly links against the system libgfortran library rather than the one provided by the `libgfortran` conda package.  I'm working on a new `openblas` package to fix this issue.  

A work around until the new package is ready is to install libgfortran using the system package manager.  `sudo apt install libgfortran3` should do the trick.

bonehead, 2017-08-15:
Installing libgfortran3 did the trick, numpy installed successfully.

jjhelmus, 2017-08-16:
I created a new `libgfortran` package for armv6l.  The old one was missing the libraries.  Updating to this new build should fix the original issue without the need to install libgfortran3 with apt.

closed 2017-08-16 by jjhelmus