jjhelmus/berryconda #15: QT5/PyQT/PyOpenGL

kmuehlbauer, 2017-09-03:
I'm wondering if it is possible to get Qt5, PyQt and PyOpenGL recipes ready.

There are some suggestions out there to compile Qt5 natively on raspberry PI 2&3. I would give it a try, but would need some guidance in which dependent packages need to come with berryconda and which are considered system-packages (coming with raspbian). The iteration cycle will be extremely slow, considering the approx. 2 days compilation time for Qt5 on an overclocked PI3 (https://wiki.qt.io/Native_Build_of_Qt5_on_a_Raspberry_Pi)

@jjhelmus In the end I would only need PyQt and PyOpenGL. So sharing your ideas would be very much welcome.

jjhelmus, 2017-09-04:
Building Qt5 on the Raspberry Pi 3 would be an involved process.  Even building the software from source on a powerful x86 machine can take a significant amount of time.

Perhaps a simpler solution would be to install Qt5 via `apt get` and build PyQt5 against this version.  This is not ideal since it depends on more libraries to be provided by the system but it is also more obtainable.  

PyOpenGL, [at least the one from conda-forge](https://github.com/conda-forge/pyopengl-feedstock),  seems to depend on `mesa-libGL-devel` being installed on the system.  I think this too would be a practical way to provide this package, in fact it may be the only way as pyopengl seems to be a pure Python package that uses ctypes to interface with an opengl library.

jjhelmus, 2017-09-05:
I created a recipe for pyopengl and uploaded packages for armv7l.  Packages for avmv6l are one their way as well.

I was able to get a PyQt build working using the Qt5 libraries provided by the apt install.  I had to disable to QtWebKitWidgets module to get the build working.  I am not familiar enough with Qt/PyQt to know how critical this option is to PyQt use.  The package was linked against many of the system libraries.

kmuehlbauer, 2017-09-05:
> I was able to get a PyQt build working using the Qt5 libraries provided by the apt install. I had to disable to QtWebKitWidgets module to get the build working. I am not familiar enough with Qt/PyQt [...]

Did you also use the conda-forge recipe? I'm a bit at loss atm from where to start.

kmuehlbauer, 2017-09-05:
> I created a recipe for pyopengl and uploaded packages for armv7l. Packages for avmv6l are one their way as well.

@jjhelmus Thanks for quickly looking into this and providing a solution. And **berryconda** is a great improvement into getting scientific python stack on the PI.

jjhelmus, 2017-09-07:
@kmuehlbauer The recipes I used for Qt/PyQt are in the [wip_qt](https://github.com/jjhelmus/berryconda/tree/wip_qt) branch.  The Qt recipe does not yet work and the PyQT recipe uses the system installed Qt5 package.  Let me know if you make any progress on this.

kmuehlbauer, 2017-09-07:
@jjhelmus This will take some time. If I get improvements I'll come back. Thanks for setting up the branch.

kmuehlbauer, 2017-11-06:
@jjhelmus Unfortunately the application I use doesn't work well with Qt5. So I copied the [pyqt recipe](https://github.com/conda-forge/pyqt-feedstock/tree/qt4/recipe) from the qt4-branch and made some changes to run it on the PI. It `conda build` well so far. I also needed to conda build pyopengl because `conda install pyopengl` didn't return (it seemed to run forever). 

I could create a conda environment with these packages (pyopengl, pyqt, sip) and do `pip install vispy` to get vispy (and numpy) into this env. I tried to use the packaged numpy, but this wasn't detected by the `pip install vispy` process. Do you have any ideas on that?

Next step is to try to get Qt build with conda.

jjhelmus, 2017-11-08:
Qt4 is available via apt which might be a short term solution.  I expect building Qt from scratch will be a long and difficult process given that building it on traditionally platforms is challenging.  

I'm not sure why `conda install pyopengl` did not work, updating conda might help.

I do not have a good answer as to why numpy was not seen by pip when installing vispy.  Does numpy show up in `pip list`?  The numpy conda package includes a .egg-info directory which pip should be detecting.  It might be good to double check that the pip in question is coming from the conda install and from the correct environment.

flatflyfish, 2019-04-02:
Apologies as i'm a bit of a noob, but can someone help explain how to install pyqt4 on berryconda via apt-get? Once installed the berryconda version of Python doesn't seem to acknowledge the pyqt4 install.

closed 2021-01-13 by kmuehlbauer