jjhelmus/berryconda #19: Installing IDE for berryconda like spyder

hbeggs, 2017-11-08:
I've installed berryconda, and it was very successful in downloading some requisite software for spectrometer use. However, I'm looking for a script editor that I can easily modify and run code in that needs to be run through berryconda (for the spectrometer). I've tried installing spyder and python-editor, but had no luck (Packages missing in current channels). Any suggestions?
PS: I'm fairly new to all this, so please bear me! Maybe the solution is super easy.

jjhelmus, 2017-11-09:
IDLE is a minimal "IDE" that is included with Python itself.  It can be run using `idle3` in a Python 3.x environment of `idle` in Python 2.7.  IDLE is bare-bones and you might be looking for something more full featured.  A number of people use Jupyter notebooks for creating and running Python code.  Jupyter is available in berryconda after installing the `notebook` package.

Jupyter Lab is a full featured browser based development environment.  Packages do not yet exist for berryconda but I'll look into adding them.

Spyder has a large number of dependency which make it difficult to build for the Raspberry Pi.

jjhelmus, 2017-11-09:
berryconda does not provide a `wxpython` package.

jjhelmus, 2017-11-10:
`jupyterlab` is now available for berryconda.  To create a new conda environment and run the application use:

```
conda create -n jl python=3.6 jupyterlab
source activate jl
jupyter lab
```

Then point your browser at the URL.  If you want to access the Jupyter Lab instance from other machines use `jupyter lab --ip XX.XX.XX.XX` replacing XX.XX.XX.XX with the IP address of the Raspberry Pi.

closed 2017-11-10 by jjhelmus

hbeggs, 2017-11-10:
Appreciate it!