jjhelmus/berryconda #21: OPencv3 with libgtk2 bindings

Esiravegna, 2017-12-31:
Do you tink it is possible to have this package with these bindings available?
I tried to compile opencv on my own to no avail.
Thanks for all your hard work!

Esiravegna, 2017-12-31:
More data:
```
OpenCV Error: Unspecified error (The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script) in cvWaitKey, file /home/jhelmus/conda/conda-bld/opencv_1512148862464/work/modules/highgui/src/window.cpp, line 676
2017-12-31 15:00:31,351 - wand.detector - ERROR - Error detecting a wand: /home/jhelmus/conda/conda-bld/opencv_1512148862464/work/modules/highgui/src/window.cpp:676: error: (-2) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function cvWaitKey
```

lodagro, 2018-01-03:
:thumbsup:

jjhelmus, 2018-01-16:
I have not been able to determine how to build OpenCV with GTK support in a manner that is consistent with how conda packages should be distributed.  Namely, GTK and its dependency should be distributed as conda package rather than relying on the system to provide these libraries.  The OpenCV recipe in conda-forge has GUI support via Qt but this would be very challenging to replicate on the Raspberry Pi.  

I will not be working on a OpenCV build with GTK but would be happy to review a recipe that support it.

Esiravegna, 2018-01-16:
Hey @jjhelmus , thanks for your reply!
Would [this](http://www.codegists.com/code/build-opencv-3/) help?

jjhelmus, 2018-01-16:
@Esiravegna Not really.  This problem is not getting OpenCV to build with GTK, rather it is getting GTK and its dependencies compiled as a conda package.

Esiravegna, 2018-01-16:
Right, thanks again. As per the creators, [this](https://github.com/loopbio/opencv-feedstock/tree/loopbio/recipe) recipe builds against libgtk2, and it appears that it only requires to have it installed.

jjhelmus, 2018-01-16:
That recipe uses QT for the GUI not GTK.