jjhelmus/berryconda #25: Including pytorch?

martinbel, 2018-05-17:
At the moment instally pytorch (cpu support, no cuda) can only be done installing from source which takes at least 8 hours! It would be great to have a binary!

jjhelmus, 2018-05-20:
This would be a good addition to berryconda. I don't have the time to develop and test a recipe but would be happy to review a pull request with a working recipe.

gaiar, 2018-05-28:
Hi! 
Seems, that I've managed to get pytorch installing via conda on my raspberry pi 3, including all dependencies. You can check here:
https://anaconda.org/gaiar/pytorch-cpu
https://anaconda.org/gaiar/torchvision

Should I create a pull request for the recipe?

Martin-Laclaustra, 2018-12-09:
@gaiar 
Dear Gaiar,
would you mind providing in your channel (https://anaconda.org/gaiar/pytorch-cpu/files) a version for python v3.5?
I am trying to get pytorch together with tensorflow in an environment and the difficulty is that tensorflow is not available for python v3.6.
Also, did you finally provide the recipe? pytorch does not appear on rpi channel.

Thank you for your work and generosity in sharing all this!

Martin-Laclaustra, 2018-12-25:
@gaiar 
I am currently working only with pytorch, but I seem to have problems with the built-in types.
My project uses numba and llvmlite as well.

I face these errors:
Prediction script (from pretrained model in other machine):
RuntimeError: Expected tensor for argument #1 'indices' to have scalar type Long; but got CPUIntTensor instead (while checking arguments for embedding)
Training script:
Operands must be the same type, got (i32, i64)

Any advice on how to align the int type length?
Thanks in advance.