jjhelmus/berryconda #29: `illegal instruction` on conda on fresh install on raspberry pi zero w

mcscope, 2018-07-09:
Hi! Just did a fresh install of this on a Raspberry Pi Zero W and am seeing strange behavior. 
I'm on Raspbarian Stretch Lite, 2018-06-27, and I just installed Berryconda. I've attached the terminal session where I discovered the issue.

Any invocation of `conda` dies with `Illegal instruction` after about a second. 
Same with `conda info` `conda install` `conda develop` etc. 
python works, and is 3.6, though I can't tell if it's the conda version 

```pi@demozero:~ $ ./Berryconda3-2.0.0-Linux-armv7l.sh

Welcome to Berryconda3 2.0.0

In order to continue the installation process, please review the license
agreement.
Please, press ENTER to continue
>>>
Copyright (c) 2016-2017 Jonathan J. Helmus
All rights reserved.

Redistribution and use in source and binary forms, with or without
modification, are permitted provided that the following conditions are
met:

    * Redistributions of source code must retain the above copyright
       notice, this list of conditions and the following disclaimer.

    * Redistributions in binary form must reproduce the above
       copyright notice, this list of conditions and the following
       disclaimer in the documentation and/or other materials provided
       with the distribution.

    * Neither the name of the developers nor the names of any
       contributors may be used to endorse or promote products derived
       from this software without specific prior written permission.

THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
"AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
(INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
        # (ISO/IEC-6429). (Lack of such support is extremely rare, and such

Do you approve the license terms? [yes|no]
[no] >>> y
Please answer 'yes' or 'no':
>>> yes

Berryconda3 will now be installed into this location:
/home/pi/berryconda3

  - Press ENTER to confirm the location
  - Press CTRL-C to abort the installation
  - Or specify a different location below

[/home/pi/berryconda3] >>>
PREFIX=/home/pi/berryconda3

installing: python-3.6.1-0 ...
^[[B^[[B^[[B^[[B^[[B^[[B^[[BPython 3.6.1
installing: asn1crypto-0.22.0-py36_0 ...
installing: ca-certificates-2017.4.17-0 ...
installing: certifi-2017.4.17-py36_0 ...
installing: cffi-1.10.0-py36_0 ...
installing: chardet-3.0.4-py36_0 ...
installing: conda-4.3.22-py36_0 ...
installing: conda-env-2.6.0-0 ...
installing: cryptography-1.9-py36_0 ...
installing: idna-2.5-py36_0 ...
installing: libffi-3.2.1-0 ...
installing: ncurses-5.9-0 ...
installing: openssl-1.0.2l-0 ...
installing: packaging-16.8-py36_0 ...
installing: pip-9.0.1-py36_0 ...
installing: pycosat-0.6.1-py36_0 ...
installing: pycparser-2.17-py36_0 ...
installing: pyopenssl-16.2.0-py36_0 ...
installing: pyparsing-2.2.0-py36_0 ...
installing: readline-6.3-0 ...
installing: requests-2.18.1-py36_0 ...
installing: ruamel_yaml-0.11.14-py36_0 ...
installing: setuptools-36.0.1-py36_0 ...
installing: six-1.10.0-py36_0 ...
installing: sqlite-3.15.0-0 ...
installing: tk-8.6.6-0 ...
installing: urllib3-1.21.1-py36_0 ...
installing: wheel-0.29.0-py36_0 ...
installing: xz-5.2.2-0 ...
installing: yaml-0.1.6-0 ...
installing: zlib-1.2.11-0 ...
installation finished.
Do you wish the installer to prepend the Berryconda3 install location
to PATH in your /home/pi/.bashrc ? [yes|no]
[no] >>>
You may wish to edit your .bashrc or prepend the Berryconda3 install location:

$ export PATH=/home/pi/berryconda3/bin:$PATH

Thank you for installing Berryconda3!
pi@demozero:~ $ less /.bash_rc
/.bash_rc: No such file or directory
pi@demozero:~ $ conda
-bash: conda: command not found
pi@demozero:~ $ vim .bashrc
pi@demozero:~ $ bash
pi@demozero:~ $ conda
Illegal instruction
pi@demozero:~ $ python --version
Python 3.6.1
pi@demozero:~ $ python
Python 3.6.1 | packaged by rpi | (default, Apr 20 2017, 19:35:19)
[GCC 4.9.2] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>
```

mcscope, 2018-07-09:
Oh I think I found the issue, I used the arm7 installer instead of the arm6. It was the wrong link. 
Trying the 6, will update this ticket if it resolve it.

jjhelmus, 2018-07-09:
@mcscope Your findings are expected.  The armv7l installer will only work on the Raspberry Pi 2 or 3.  The armv6l installer should be used on the Zero and 1.