jjhelmus/berryconda #30: OpenCV with imshow

qenops, 2018-07-10:
In attempting to view images in openCV I run into this issue:

```
OpenCV Error: Unspecified error (The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script) in cvShowImage, file /home/jhelmus/conda/conda-bld/opencv_1512148862464/work/modules/highgui/src/window.cpp, line 605
Traceback (most recent call last):
  File "stereoCam.py", line 18, in <module>
    cv2.imshow('Camera', image)
cv2.error: /home/jhelmus/conda/conda-bld/opencv_1512148862464/work/modules/highgui/src/window.cpp:605: error: (-2) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function cvShowImage
```

So it looks like openCV wasn't compiled with UI support.  Is this something that you could provide?

For completeness this is the code snippet that generates the error:
```
from picamera.array import PiRGBArray
from picamera import PiCamera
import time
import cv2

camera = PiCamera()
camera.resolution = (640, 480)
camera.framerate = 30
rawCapture = PiRGBArray(camera, size=camera.resolution)

# pause while the camera inits
time.sleep(0.1)

for frame in camera.capture_continuous(rawCapture, format='bgr', use_video_port=True):
    image = frame.array
    # display using opencv
    cv2.imshow('Camera', image)
    key = cv2.waitKey(1) & 0xFF
    # clear the stream for the next frame
    rawCapture.truncate(0)
    if key == ord('q'):
        break

```

vamshireddyp, 2018-07-12:
can you first try checking 
$ python
>>> import cv2

jjhelmus, 2018-07-12:
Building a version of OpenCV with a GUI backend is difficult.  See #21

closed 2018-07-12 by qenops