jjhelmus/berryconda #35: install Tensorflow

DaliaAshqar, 2018-07-29:
I have installed  tensorflow using conda and it did not show any error, but when I tried to check if the installation works properly it shows the following error:
no module named tensorflow

jjhelmus, 2018-07-29:
How did you install Tensorflow?  The `rpi` channel does not contain a tensorflow package.

DaliaAshqar, 2018-07-29:
Firstly I searched for It using the following command:
              anaconda search -t conda tensorflow 
then It displayed for me a list of the available packages, I choose the one which is suitable for Raspberry pi 3 platform, then I Run the following command:
              anaconda show <package name> 
to guide me how to install it 
              conda install --channel <package name/ channel >.
And to make sure it was installed I typed
              conda list 
and tensorflow was appearing in the list, however; when I try to use import tensorflow it showed me an error. Then I realized that there is something wrong, so I installed using pip.

So, maybe using "conda list" to check whether the package is installed or not is not enough.
Thank you so much for your replay.

gaiar, 2018-07-31:
Currently getting tensorflow via conda for ARM devices is challenging. I would recommend you following next steps

1. Create Python 3.5 environment as `conda create -n YOUR_NAME python=3.5` 
2. Download suitable for you version from  https://github.com/lhelontra/tensorflow-on-arm/releases
3. Install it via `pip` in your environment

crawforc3, 2018-08-04:
Tensorflow is officially supported for rpi's now. Installing should be a easy as `pip install tensorflow`

https://medium.com/tensorflow/tensorflow-1-9-officially-supports-the-raspberry-pi-b91669b0aa0
https://www.tensorflow.org/install/install_raspbian