jjhelmus/berryconda #48: ERROR: size of Berryconda2-2.0.0-Linux-armv7l.sh should be     33565270 bytes

RenewedMe, 2018-09-23:
Hi,

I downloaded the file for Python 2 on a RaspberryPi 3B+ on my Mac and then used FTP to put it on the raspberry pi. I then made the .sh file executable using instructions provided but when I then try to execute it I get the following message:

ERROR: size of Berryconda2-2.0.0-Linux-armv7l.sh should be     33565270 bytes

Any idea what I a doing wrong?

llarouche, 2019-11-01:
Did you transfer in binary mode? Do you have the same error with scp? Maybe you had an incomplete download from the internet.

Bulldozzer82, 2020-02-12:
I had the similar issue. I solve the problem using wget
`wget https://github.com/jjhelmus/berryconda/releases/download/v2.0.0/Berryconda3-2.0.0-Linux-armv6l.sh'
'chmod +x Berryconda3-2.0.0-Linux-armv6l.sh.1`
'./Berryconda3-2.0.0-Linux-armv6l.sh.1'

But check if the file name hadn't been change as it was in my case due to multiple copies. After wget download will be completed you get a message: 
`Saving to: 'Berryconda3-2.0.0-Linux-armv6l.sh**.1**` use exactly this filename further.