jjhelmus/berryconda #51: JupyterLab update

betteridiot, 2018-11-06:
As of right now, Jupyter Lab on the rpi channel is stuck in a [pre-alpha version of a computational environment for Jupyter](https://anaconda.org/rpi/jupyterlab). Any chance an updated build could be done?

jjhelmus, 2018-11-07:
An updated version of the jupyterlab package would be good.  

JupyterLab version 0.34 and above requires nodejs to build, conda-forge/jupyterlab-feedstock#128.  I'm running into some issues getting the conda-forge [nodejs recipe](https://github.com/conda-forge/nodejs-feedstock) to build on my Raspberry Pi 3+.  If anyone is able to get this working please submit a PR with the recipe.

jjhelmus, 2018-11-11:
I was able to build nodejs by disabling parallel builds in make.  Jupyterlab version 0.35.4 is now available in the `rpi` channel

closed 2018-11-11 by jjhelmus

betteridiot, 2018-11-12:
Thank you so much. My RPi3B+ has become my default machine for teaching the classroom. Your channel has become the fulcrum in leveraging the technology. Thank you.