jjhelmus/berryconda #59: Issue with jupyter notebook

ModMaamari, 2019-02-12:
When I try to run :
`jupyter notebook
`
I get this error :

```
Traceback (most recent call last):
  File "/home/pi/berryconda3/bin/jupyter-notebook", line 7, in <module>
    from notebook.notebookapp import main
  File "/home/pi/berryconda3/lib/python3.6/site-packages/notebook/notebookapp.py", line 46, in <module>
    from zmq.eventloop import ioloop
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/__init__.py", line 47, in <module>
    from zmq import backend
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/__init__.py", line 40, in <module>
    reraise(*exc_info)
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/utils/sixcerpt.py", line 34, in reraise
    raise value
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/__init__.py", line 27, in <module>
    _ns = select_backend(first)
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/select.py", line 27, in select_backend
    mod = __import__(name, fromlist=public_api)
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/cython/__init__.py", line 6, in <module>
    from . import (constants, error, message, context,
ImportError: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/cython/../../../../../libzmq.so.5)

```

I installed jupyter notebook using conda, and I checked that it is installed and it is in (conda list)

How can I run jupyter notebook ?

jjhelmus, 2019-02-12:
@ModMaamari What version of Rasbian are you running?  You may need to update to a newer version.

ModMaamari, 2019-02-12:
>> `uname -a`
`Linux pi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux`

ModMaamari, 2019-02-12:
Is the cause of the error is the version of my OS?
HELP PLEASE !!

FYI :
ipython is working without any errors, but jupyter is not wrking.
`>>ipython
Python 3.6.1 | packaged by rpi | (default, Apr 20 2017, 19:35:19) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.5.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: 
`
`>>uname -a
Linux pi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux`
_______________________________________________________________________________________________________
`>>cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"`
_______________________________________________________________________________________________________
`>>conda list
# packages in environment at /home/pi/berryconda3:
#
# Name                    Version                   Build  Channel
asn1crypto                0.22.0                   py36_0  
backcall                  0.1.0                      py_0  
backports                 1.0                        py_2  
backports.functools_lru_cache 1.5                        py_1  
blas                      1.1                    openblas  
bleach                    2.1.4                      py_1  
ca-certificates           2018.8.24                     0  
certifi                   2018.8.24                py36_1  
cffi                      1.10.0                   py36_0  
chardet                   3.0.4                    py36_0  
conda                     4.5.11                   py36_0  
conda-env                 2.6.0                         0  
cryptography              1.9                      py36_0  
cycler                    0.10.0                     py_1  
cython                    0.28.5           py36h4f752ac_0  
decorator                 4.3.0                      py_0  
entrypoints               0.2.3                    py36_2  
freetype                  2.7                           0  
html5lib                  1.0.1            py36h4e0ed57_0  
idna                      2.5                      py36_0  
ipykernel                 4.9.0                    py36_0  
ipython                   6.5.0                    py36_0  
ipython_genutils          0.2.0                      py_1  
ipywidgets                7.4.1                      py_0  
jedi                      0.12.1                   py36_0  
jinja2                    2.10                       py_1  
jsonschema                2.6.0                    py36_2  
jupyter                   1.0.0            py36h224aed1_0  
jupyter_client            5.2.3                      py_1  
jupyter_console           5.2.0                    py36_1  
jupyter_core              4.4.0              pyhc2c3be9_0  
libffi                    3.2.1                         0  
libgfortran               3.0.0                         0  
libpng                    1.6.35               h849d6a0_1  
libsodium                 1.0.16                        0  
markupsafe                1.0              py36hdff2a78_1  
matplotlib                2.1.2            py36hdcec099_0  
mistune                   0.8.3            py36hdff2a78_2  
nbconvert                 5.3.1                      py_1  
nbformat                  4.4.0                      py_1  
[... 77 lines cut ...]
/home/pi
`_______________________________________________________________________________________________________
`>>jupyter notebook
Traceback (most recent call last):
  File "/home/pi/berryconda3/bin/jupyter-notebook", line 7, in <module>
    from notebook.notebookapp import main
  File "/home/pi/berryconda3/lib/python3.6/site-packages/notebook/notebookapp.py", line 46, in <module>
    from zmq.eventloop import ioloop
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/__init__.py", line 47, in <module>
    from zmq import backend
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/__init__.py", line 40, in <module>
    reraise(*exc_info)
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/utils/sixcerpt.py", line 34, in reraise
    raise value
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/__init__.py", line 27, in <module>
    _ns = select_backend(first)
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/select.py", line 27, in select_backend
    mod = __import__(name, fromlist=public_api)
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/cython/__init__.py", line 6, in <module>
    from . import (constants, error, message, context,
ImportError: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/cython/../../../../../libzmq.so.5)
`
_______________________________________________________________________________________________________

PLEASE HELP!

jjhelmus, 2019-02-12:
> VERSION="8 (jessie)"

You need to update to stretch to use the berryconda packages.  Older packages were built to support jessie but newer packages require stretch.

ModMaamari, 2019-02-12:
Thanks, I'll try to update it, hope that solves the problem :)