jjhelmus/berryconda #64: openssl error message after running 'conda update --all'

carver7, 2019-03-31:
<loginname>@<raspberryPisName>:~ $ openssl
WARNING: can't open config file: /home/jhelmus/bc3_v7/conda-bld/openssl_1547610726478/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_place/ssl/openssl.cnf

jjhelmus, 2019-04-13:
A prefix in the openssl `libcrypto` library was not getting marked for replacement.  I'll upload correct packages shortly.

closed 2019-04-13 by jjhelmus

jjhelmus, 2019-04-13:
The openssl 1.0.2r packages now available on the `rpi` channel have a fix for this issue:

```
~$ openssl version
OpenSSL 1.0.2r  26 Feb 2019
~$ openssl
OpenSSL>  
```

commit 3df3a1b8 (2019-04-13): update openssl recipe to 1.0.2r

Remove ignore_prefix_files

closes #64