jjhelmus/berryconda #66: installing jupyter notebook and lab in berryconda

ludiusvox, 2019-04-11:
I souced Conda and installed jupyter it seems to install correctly but I am getting this error:

Traceback (most recent call last):
  File "/home/pi/berryconda3/bin/jupyter-notebook", line 7, in <module>
    from notebook.notebookapp import main
  File "/home/pi/berryconda3/lib/python3.6/site-packages/notebook/notebookapp.py", line 46, in <module>
    from zmq.eventloop import ioloop
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/__init__.py", line 47, in <module>
    from zmq import backend
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/__init__.py", line 40, in <module>
    reraise(*exc_info)
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/utils/sixcerpt.py", line 34, in reraise
    raise value
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/__init__.py", line 27, in <module>
    _ns = select_backend(first)
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/select.py", line 27, in select_backend
    mod = __import__(name, fromlist=public_api)
  File "/home/pi/berryconda3/lib/python3.6/site-packages/zmq/backend/cython/__init__.py", line 6, in <module>
    from . import (constants, error, message, context,
ImportError: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /home/pi/berrycon

Possibly jupyter isn't supported by ARM processors or is there another pre-requisite library I need to install as a dependancy?

jjhelmus, 2019-04-12:
What version of Raspbian do you have installed?  Berryconda requires at Stretch.

ludiusvox, 2019-04-12:
yeah I caught that I am working on installing stretch.  Thanks for your
reply.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ludiusvox, 2019-04-13:
here is the instructions for installing stretch, I think the major vendors aren't selling raspberry pi with stretch installed yet.

https://linuxconfig.org/how-to-upgrade-debian-8-jessie-to-debian-9-stretch

ludiusvox, 2019-04-13:
These instructions are correct but Pi breaks, the only way to install to stretch to burn from rasberry pi england website.

jjhelmus, 2019-04-16:
This is not an issue with Berryconda.  If you require help installing Raspbian please ask in a forum for the distribution.

closed 2019-04-16 by jjhelmus

ludiusvox, 2019-04-17:
Ok thanks