jjhelmus/berryconda #67: pip upgraded: matplotlib, pandas, numpy

ludiusvox, 2019-04-14:
I have only berryconda 3 installed, I used 'conda install matplotlib', it works successfully, but when I go to run it in Jupyter notebook it doesn't compile it returns an error 'library not found'.  This error occurs for pandas and matplotlib in jupyter.

I flashed installed the newest version of debian and that's how I got matplot lib working.
I am not getting these same errors in jupyter on ubuntu on my PC.

List of commands not working in jupyter:
%matplotlib inline
import pandas as pd
import numpy as np

I saw the youtube video supposedly scientific python is supposed to come with berryconda so I don't know what the deal is

jjhelmus, 2019-04-16:
Did you use conda to install notebook, numpy, and pandas?  Berryconda creates a separate install of Python from the system provided version.  You need to install the libraries you require into this environment.  If you do not conda install notebook the system provided version will be used.  You can verify which version will be run by looking at the result of `which jupyter`.  The berryconda version should be first in the list and begin with the path where you installed berryconda.

ludiusvox, 2019-04-17:
Ok I will check it .  I might not have created the environment correctly
and just used the “base” environment.  Let me try troubleshooting it further