jjhelmus/berryconda #69: Buster released. Things are not working.

greenpdx, 2019-07-22:
Hi Buster has been released with python 3.7.3  I can install berryconda but when I try to install all the other modules needed things break apart. I have built  opencv 4.1.0 for python 3.7.
There are three ways to install the other modules
sudo apt, pip3 install and conda install.
the pip3 install takes forever
I install everything using apt install but that does not work for python 3.6
I am teaching a class using RasPi and I want to teach jupyter notbook from the RPi.
Any ideas on how to get jupyter working on buster?

greenpdx, 2019-07-28:
Ok,  This is what I did.  I build python3.6.1 and installed it at /usr/local/p36, created a ld.so.cong.d/p36.conf to point to the library.  put /usr/local/p36/bin at the front of the PATH.
Now install Berryconda3-2.0.0-Linux-armv7l.sh -b
I do have to add $HOME/berryconda3/bin to the front of the PATH.
pip install --upgrade pip to upgrade it to v19.2.

because I am using pip inside berryconda 
I can use pip install and it is installed in the berryconda path.

I also built opencv 4.1.0 for python3.6 for the conda

jjhelmus, 2019-07-30:
berryconda only provides Python 3.6, so if you go that route you will need to use that version.  AFAIK there should be no issues with berryconda on buster.

greenpdx, 2019-07-31:
The issues I have is I needed to install opencv 4.1.0 and because berryconda uses 3.6.1 and buster uses 3.7.3 I had issues building cv2. Yes you are correct berryconda does not have issues, but installing all the ML python modules was painful and slow. Some took hours and hours to build.  I wanted to use apt to install them. Look at https://github.com/greenpdx/SAAIBase

closed 2019-07-31 by greenpdx