jjhelmus/berryconda #71: Create an installer for RPI 4

malynda, 2019-08-09:
Create an installer for Armv8
(so far the v7 installer seems to work okay, as I installed it about 2 minutes ago)

jjhelmus, 2019-08-09:
What Linux distribution are you running on your Raspberry Pi 4?  Raspbian is 32-bit so packages built for aarch64 will not run on such a system.  

[conda-forge](https://conda-forge.org/) has a large collection of linux-aarch64 packages which can be used on systems that are capable of running this binaries.

malynda, 2019-08-09:
Raspbian Buster, which is still 32 bit, and I assume why the v7 installer worked okay.

jjhelmus, 2019-08-09:
I suspect the v7 installers will work fine on a 4, but I do not have one to test.  The RPi 3 is a armv8 processor but when run on Raspian `uname -a` would report armv7l.

malynda, 2019-08-09:
Indeed it does:

$ uname -a
Linux raspberrypi 4.19.58-v7l+ #1245 SMP Fri Jul 12 17:31:45 BST 2019 armv7l GNU/Linux

closed 2019-08-09 by malynda