jjhelmus/berryconda #72: libpython3.6m.so.1 is missing

tadam98, 2019-10-13:
Hi,

Your installation of python 3.6.1 is without the shared library resource in:
/usr/lib/arm-linux-gnueabihf/libpython3.6m.so.1

[https://github.com/pyenv/pyenv/issues/917](url) says that BEFORE installing python the following should be set:
CONFIGURE_OPTS=--enable-shared pyenv install 3.6.1

Without this, cmake of OpenCV 3.0.0 fails for python 3.6.1, it does not find the library which is a must for numpy.

follow:
[https://www.pyimagesearch.com/2017/09/04/raspbian-stretch-install-opencv-3-python-on-your-raspberry-pi/](url)

Can you kindly check this up ?

Best,
Mickey

===================================================
I am using Raspberry Pi 4 model B:
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"

jjhelmus, 2019-10-15:
I do not understand how this issue is with berryconda. Can you provide more details about how you have installed berryconda and what you are trying to accomplish?

jjhelmus, 2019-10-16:
The libpython3.6m.so.1 file is missing from the python 3.6 packages in berryconda.  The package does supply `libpython3.so`, `libpython3.6m.so` and `libpython3.6m.so.1.0` files in the `lib` directory of the environment where the package is install.  A symlink the the last of these would be the appropriate target for the `.so.1` file.  In most cases the un-numbered SO will suffice.  

When building OpenCV the Python library can be passed during configuration with `-DPYTHON_LIBRARY=/path/to/libpython3.6.m.so`.  The [opencv-feedstock](https://github.com/conda-forge/opencv-feedstock) may offer additional hints on building OpenCV against Python not installed in the system location.

tadam98, 2019-10-16:
Hi

I intalled it twice on a clean Raspberry Pi 4B. In both case neither libpython3.6m.so<http://libpython3.6m.so/> nor libpython3.6m.so<http://libpython3.6m.so/>.1.0 were
installed.

So, again on a clean OS, I installed python 3.6.1 and the dev package of 3.6.1  BEFORE   installing berryconda. And the make file detected everything without a need for:

DPYTHON_LIBRARY=/path/to/libpython3.6.m.so

The suggested DPYTHON_LIBRARY would solve the problem of OpenCV that looks at the python version of conda instead of the version I install under an active environment under which I would want to compile OpenCV.

Best,
Mickey

jjhelmus, 2019-10-16:
> I intalled it twice on a clean Raspberry Pi 4B. In both case neither libpython3.6m.so<http://libpython3.6m.so/> nor libpython3.6m.so<http://libpython3.6m.so/>.1.0 were installed.

Where are you looking for these files?  They should be in the `lib` directory under the folder where berryconda is installed.  I just did a fresh install and all three files were present.

tadam98, 2019-10-16:
Nope. I looked in the folder I wrote. Will check again. And if there I will use the suggeted define.

I amny case, I would prefer the standard installtion onto the folder I wrote as your install should be "under" the conda rather than one of the environments.

Unless there is a python path pointing to your folder after the installation which OpenCV would use. This did not happen for me.

Best,
Mickey

jjhelmus, 2019-10-16:
> I amny case, I would prefer the standard installtion onto the folder I wrote as your install should be "under" the conda rather than one of the environments.

Conda is designed to install files into a location separate from the operating system paths.  Installing into system paths is not possible.

tadam98, 2019-10-16:
Will try again on a fresh install and let you know.

Best,
Mickey

tadam98, 2019-10-18:
Hi,

Adding the suggested -D worked fine.
(Compilation of OpenCV on RP4B fails because of missing sys/videoio.h but this is not a conda isssue. Thank you for your quick response).

Best,
Mickey

closed 2019-10-18 by tadam98