jjhelmus/berryconda #73: OpenCV 4.1.1 - Successful procedure

tadam98, 2019-10-19:
Hi,

I have successfully compiled and installed OpenCV 4.1.1 for both Python2 and Python3 environments.
1. Follow: https://qengineering.eu/install-opencv-4.1-on-raspberry-pi-4.html
2. The paths of the above procedure need to be **updated for BerryConda** as follows:
    -D PYTHON3_LIBRARY=/home/pi/berryconda3/lib/libpython3.6m.so \
    -D PYTHON3_PACKAGES_PATH=/home/pi/berryconda3/lib/python3.6/site-packages \
    -D BUILD_opencv_python3=TRUE \
    -D BUILD_opencv_python2=YES \
[CMAKE-OpenCV-4.1.1.txt](https://github.com/jjhelmus/berryconda/files/3746993/CMAKE-OpenCV-4.1.1.txt)
[image]

Notes: Unlike all other OpenCV procedures I tried - they all miss **sys/videoio.h** - this one works well.

Firemanpl, 2020-07-03:
how ??? I can't do that.
`pi@raspberrypi:~/opencv/build $ cmake -D CMAKE_BUILD_TYPE=RELEASE \
>     -D CMAKE_INSTALL_PREFIX=/usr/local \
>     -D OPENCV_EXTRA_MODULES_PATH=~/opencv_contrib/modules \
>     -D PYTHON3_LIBRARY=/home/pi/berryconda3/lib/libpython3.6m.so \
>     -D PYTHON3_PACKAGES_PATH=/home/pi/berryconda3/lib/python3.6/site-packages \
>     -D BUILD_opencv_python3=TRUE \
>     -D BUILD_opencv_python2=YES \
>     -D ENABLE_NEON=ON \
>     -D ENABLE_VFPV3=ON \
>     -D BUILD_TIFF=ON \
>     -D WITH_FFMPEG=ON \
>     -D WITH_GSTREAMER=ON \
>     -D WITH_TBB=ON \
>     -D BUILD_TBB=ON \
>     -D BUILD_TESTS=OFF \
>     -D WITH_EIGEN=OFF \
>     -D WITH_V4L=ON \
>     -D WITH_LIBV4L=ON \
>     -D WITH_VTK=OFF \
>     -D CMAKE_SHARED_LINKER_FLAGS=-latomic \
>     -D OPENCV_ENABLE_NONFREE=ON \
>     -D INSTALL_C_EXAMPLES=OFF \
>     -D INSTALL_PYTHON_EXAMPLES=OFF \
>     -D BUILD_NEW_PYTHON_SUPPORT=ON \
>     -D BUILD_opencv_python3=TRUE \
>     -D OPENCV_GENERATE_PKGCONFIG=ON \
>     -D BUILD_EXAMPLES=OFF ..
-- Detected processor: armv7l
-- Looking for ccache - not found
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found suitable version "1.2.11", minimum required is "1.2.3") 
-- Could NOT find OpenJPEG (minimal suitable version: 2.0, recommended version >= 2.3.1)
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found version "1.2.11") 
-- Could not find OpenBLAS include. Turning OpenBLAS_FOUND off
-- Could not find OpenBLAS lib. Turning OpenBLAS_FOUND off
-- Could NOT find Atlas (missing: Atlas_CLAPACK_INCLUDE_DIR) 
-- A library with LAPACK API found.
-- Could NOT find JNI (missing: JAVA_AWT_LIBRARY JAVA_JVM_LIBRARY JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2 JAVA_AWT_INCLUDE_PATH) 
-- OpenCV Python: during development append to PYTHONPATH: /home/pi/opencv/build/python_loader
-- Checking for module 'libavresample'
--   No package 'libavresample' found
-- Checking for module 'gstreamer-base-1.0'
--   No package 'gstreamer-base-1.0' found
-- Checking for module 'gstreamer-app-1.0'
--   No package 'gstreamer-app-1.0' found
-- Checking for module 'gstreamer-riff-1.0'
--   No package 'gstreamer-riff-1.0' found
-- Checking for module 'gstreamer-pbutils-1.0'
--   No package 'gstreamer-pbutils-1.0' found
-- Checking for module 'libdc1394-2'
--   No package 'libdc1394-2' found
-- Module opencv_alphamat disabled because the following dependencies are not found: Eigen
-- Caffe:   NO
-- Protobuf:   NO
-- Glog:   NO
-- freetype2:   YES (ver 22.1.16)
-- harfbuzz:    YES (ver 2.3.1)
-- HDF5: Using hdf5 compiler wrapper to determine C configuration
-- Module opencv_ovis disabled because OGRE3D was not found
-- No preference for use of exported gflags CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported gflags CMake configuration if available.
-- Failed to find installed gflags CMake configuration, searching for gflags build directories exported with CMake.
-- Failed to find gflags - Failed to find an installed/exported CMake configuration for gflags, will perform search for installed gflags components.
-- Failed to find gflags - Could not find gflags include directory, set GFLAGS_INCLUDE_DIR to directory containing gflags/gflags.h
-- Failed to find glog - Could not find glog include directory, set GLOG_INCLUDE_DIR to directory containing glog/logging.h
-- Module opencv_sfm disabled because the following dependencies are not found: Eigen Glog/Gflags
-- Checking for module 'tesseract'
--   No package 'tesseract' found
-- Tesseract:   NO
-- Allocator metrics storage type: 'int'
-- HDF5: Using hdf5 compiler wrapper to determine C configuration
[... 101 lines cut ...]
--     Libraries:                   /usr/lib/arm-linux-gnueabihf/libpython2.7.so (ver 2.7.16)
--     numpy:                       /usr/lib/python2.7/dist-packages/numpy/core/include (ver 1.16.2)
--     install path:                lib/python2.7/dist-packages/cv2/python-2.7
-- 
--   Python 3:
--     Interpreter:                 /home/pi/berryconda3/bin/python3 (ver 3.6.1)
--     Libraries:                   NO
--     numpy:                       NO (Python3 wrappers can not be generated)
--     install path:                -
-- 
--   Python (for build):            /usr/bin/python2.7
-- 
--   Java:                          
--     ant:                         NO
--     JNI:                         NO
--     Java wrappers:               NO
--     Java tests:                  NO
-- 
--   Install to:                    /usr/local
-- -----------------------------------------------------------------
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/opencv/build
`

Firemanpl, 2020-07-03:
How??? Idan't do that.
```
pi@raspberrypi:~/opencv/build $ cmake -D CMAKE_BUILD_TYPE=RELEASE \
>     -D CMAKE_INSTALL_PREFIX=/usr/local \
>     -D OPENCV_EXTRA_MODULES_PATH=~/opencv_contrib/modules \
>     -D PYTHON3_LIBRARY=/home/pi/berryconda3/lib/libpython3.6m.so \
>     -D PYTHON3_PACKAGES_PATH=/home/pi/berryconda3/lib/python3.6/site-packages \
>     -D BUILD_opencv_python3=TRUE \
>     -D BUILD_opencv_python2=YES \
>     -D ENABLE_NEON=ON \
>     -D ENABLE_VFPV3=ON \
>     -D BUILD_TIFF=ON \
>     -D WITH_FFMPEG=ON \
>     -D WITH_GSTREAMER=ON \
>     -D WITH_TBB=ON \
>     -D BUILD_TBB=ON \
>     -D BUILD_TESTS=OFF \
>     -D WITH_EIGEN=OFF \
>     -D WITH_V4L=ON \
>     -D WITH_LIBV4L=ON \
>     -D WITH_VTK=OFF \
>     -D CMAKE_SHARED_LINKER_FLAGS=-latomic \
>     -D OPENCV_ENABLE_NONFREE=ON \
>     -D INSTALL_C_EXAMPLES=OFF \
>     -D INSTALL_PYTHON_EXAMPLES=OFF \
>     -D BUILD_NEW_PYTHON_SUPPORT=ON \
>     -D BUILD_opencv_python3=TRUE \
>     -D OPENCV_GENERATE_PKGCONFIG=ON \
>     -D BUILD_EXAMPLES=OFF ..
-- Detected processor: armv7l
-- Looking for ccache - not found
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found suitable version "1.2.11", minimum required is "1.2.3") 
-- Could NOT find OpenJPEG (minimal suitable version: 2.0, recommended version >= 2.3.1)
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found version "1.2.11") 
-- Could not find OpenBLAS include. Turning OpenBLAS_FOUND off
-- Could not find OpenBLAS lib. Turning OpenBLAS_FOUND off
-- Could NOT find Atlas (missing: Atlas_CLAPACK_INCLUDE_DIR) 
-- A library with LAPACK API found.
-- Could NOT find JNI (missing: JAVA_AWT_LIBRARY JAVA_JVM_LIBRARY JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2 JAVA_AWT_INCLUDE_PATH) 
-- OpenCV Python: during development append to PYTHONPATH: /home/pi/opencv/build/python_loader
-- Checking for module 'libavresample'
--   No package 'libavresample' found
-- Checking for module 'gstreamer-base-1.0'
--   No package 'gstreamer-base-1.0' found
-- Checking for module 'gstreamer-app-1.0'
--   No package 'gstreamer-app-1.0' found
-- Checking for module 'gstreamer-riff-1.0'
--   No package 'gstreamer-riff-1.0' found
-- Checking for module 'gstreamer-pbutils-1.0'
--   No package 'gstreamer-pbutils-1.0' found
-- Checking for module 'libdc1394-2'
--   No package 'libdc1394-2' found
-- Module opencv_alphamat disabled because the following dependencies are not found: Eigen
-- Caffe:   NO
-- Protobuf:   NO
-- Glog:   NO
-- freetype2:   YES (ver 22.1.16)
-- harfbuzz:    YES (ver 2.3.1)
-- HDF5: Using hdf5 compiler wrapper to determine C configuration
-- Module opencv_ovis disabled because OGRE3D was not found
-- No preference for use of exported gflags CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported gflags CMake configuration if available.
-- Failed to find installed gflags CMake configuration, searching for gflags build directories exported with CMake.
-- Failed to find gflags - Failed to find an installed/exported CMake configuration for gflags, will perform search for installed gflags components.
-- Failed to find gflags - Could not find gflags include directory, set GFLAGS_INCLUDE_DIR to directory containing gflags/gflags.h
-- Failed to find glog - Could not find glog include directory, set GLOG_INCLUDE_DIR to directory containing glog/logging.h
-- Module opencv_sfm disabled because the following dependencies are not found: Eigen Glog/Gflags
-- Checking for module 'tesseract'
--   No package 'tesseract' found
-- Tesseract:   NO
-- Allocator metrics storage type: 'int'
[... 103 lines cut ...]
--     numpy:                       /usr/lib/python2.7/dist-packages/numpy/core/include (ver 1.16.2)
--     install path:                lib/python2.7/dist-packages/cv2/python-2.7
-- 
--   Python 3:
--     Interpreter:                 /home/pi/berryconda3/bin/python3 (ver 3.6.1)
--     Libraries:                   NO
--     numpy:                       NO (Python3 wrappers can not be generated)
--     install path:                -
-- 
--   Python (for build):            /usr/bin/python2.7
-- 
--   Java:                          
--     ant:                         NO
--     JNI:                         NO
--     Java wrappers:               NO
--     Java tests:                  NO
-- 
--   Install to:                    /usr/local
-- -----------------------------------------------------------------
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/opencv/build

```

Firemanpl, 2020-07-03:
I used with newest version. [link](https://qengineering.eu/install-opencv-4.3-on-raspberry-pi-4.html)