jjhelmus/berryconda #74: Berryconda is not able to be launched.

funfwo, 2019-10-24:
I once tried to install berryconda on Pi4. I installed and remove Miniconda before installing Berry. After installing Berry, I cannot use conda, since it shows -bash: /home/pi/miniconda3/bin/conda: no such file or directory.
But I have changed .bashrc when installing berry to pi/berryconda
What problem may lay here?Thanks

jjhelmus, 2019-10-25:
You will need to modify your .bashrc file to add the `bin` directory under the berryconda install to your path.  You should remove any changes that Miniconda made at the same time.

funfwo, 2019-10-25:
Thanks, But I have uninstalled Miniconda. How can I find out what new lines are added by Miniconda.

jjhelmus, 2019-10-25:
There should be lines which modify the PATH variable.  They likely are at or near the end of the .bashrc or .profile file.

funfwo, 2019-10-25:
I will check it tomorrow. Thank you.