jjhelmus/berryconda #81: Issue with conda version

step21, 2020-02-18:
In the constructor, you set version as 2.0.0, while conda still reports its version more accurately as 4.3.22. This makes it impossible to install conda-build (and maybe others) as far as I can tell, as the version picked up is 2.0.0 which is obviously not high enough. So why not more or less match the conda versioning, assuming versions are relatively compatible?

jjhelmus, 2020-03-25:
The version in the construct.yaml file set the version of the installer, e.g. Berryconda3-**2.0.0**-Linux-armv7l.sh.  The conda version is independent from this version.  

The Miniconda installers provided by Anaconda, Inc are created so that the installer version matches the conda version but this is not a requirement just a convention.

step21, 2020-05-18:
It just seem

closed 2020-05-18 by step21