jjhelmus/berryconda #82: Any statement about RPi4 and arm8?

deeplook, 2020-02-22:


step21, 2020-02-23:
2/3 is about python version. And there will not, as the project is no longer maintained. see here: https://github.com/jjhelmus/berryconda/issues/79#issuecomment-587539948

jjhelmus, 2020-03-24:
Conda-forge has been building 64-bit ARM packages for some time.  These will work on a Raspberry Pi 4 if a 64-bit operating system is installed, for example the [64-bit version of Ubuntu](https://ubuntu.com/download/raspberry-pi).  The [miniforge](https://github.com/conda-forge/miniforge) repository has links to installers for the linux-aarch64 platform which should be used.

The Raspberry Pi has had an ARMv8 CPU since the RPi 2 B model v1.2.  For backwards compatibility Raspbian run in 32-bit mode and the cpu is typically identified as `armv7l` or `armv7`.  Either of the linux-armv7l Berryconda installers should run on the Raspberry Pi 2, 3, or 4's.  

If the cpu is identified as `armv8l` it may be necessary to set the `CONDA_SUBDIR` environment variable to `linux-armv7l` as conda is not aware that armv8l in compatible with armv7l.

step21, 2020-03-24:
Mmmg interesting. So this would mean a separate installer and rpi channel is only needed for rpi 2/3? Or am I misunderstanding?

jjhelmus, 2020-03-25:
The `rpi` channel and installers can be used with Raspberry Pi 2 (v1.2), 3 or 4s running Raspbian.  This will use the packages in the `rpi` channel for the `linux-armv7l` platform.  

It should be possible to run a 64-bit version of Ubuntu on any of these, in which case the `linux-aarch64` miniforge installer and `conda-forge` packages should be used.

tiantian1412, 2020-07-25:
I want to install berryconda on Pi4, but error, u say 'set the CONDA_SUBDIR environment variable to linux-armv7l' ,does it effective, can u describe it on detail ? I dont understand